perdidor/Arduino-MDB-UART #12: Whats the purpose of "MDB_Master.pdf" ?

ottelo9, 2022-03-30:
I checked your pcb folder and saw the MDB_Master schematic. Only with a MAX232. Is this schematic for a simple RS232-MDB Converter cable?

I ask because I want to program my coin acceptor with the manufacturer software and via the MDB connection. So I need a simple and dump converter circuit that converts my RS232 from PC to 9Bit MDB.

perdidor, 2022-04-01:
if you mean burn firmware by "program coin acceptor" - this repository does not meet your needs.
if you want to control MDB coin acceptor/changer or bill acceptor - read main page of this repo.
Cheers.

perdidor, 2022-04-01:
MDB_Master schematic file was used as reference while developing this adapter

closed 2022-04-02 by perdidor